tom-2015/rpi-ws2812-server #1: Create a second channel

SmashedBird, 2015-11-06:
I can't create a second channel.
I tried :
setup channels=2
setup channel_1_gpio=18 
render // works
setup channel_2_gpio=23
Initialization failed.
: Invalid argument

tom-2015, 2015-11-07:
will take a look at it asap

tom-2015, 2015-11-08:
You can only use GPIO 12 and 18 for channel 1 and GPIO 13 and 19 for channel 2 (Requires Raspberry pi 2/B+)
Try this:

setup channels=2
setup channel_1_gpio=18
setup channel_2_gpio=13
setup channel_1_count=5
setup channel_2_count=5
rainbow 1
rainbow 2
brightness 1,32
brightness 2,32
render 1
render 2
rotate 2,1
render 2

don't forget the brightness for channel 2, it seems 0 by default

tom-2015, 2015-11-08:
I've also fixed a bug today, the setup command didn't work properly when you forgot the space after each setting.
Added 2 commands:
debug to enable debug mode
settings to view the configured settings for each channel.

SmashedBird, 2015-11-09:
Thank you very much. It works !

closed 2015-11-09 by SmashedBird