tom-2015/rpi-ws2812-server #4: Rpi Zero W

Duwacell, 2017-04-12:
Hi,

i'm trying to make it work for the Rpi Zero W but the script returns : 
"ws2811_init failed: Hardware revision is not supported"

The jgarff lib has been updated to work with Pizero W, i tried to modify your rpihw.c ( here : 
https://github.com/jgarff/rpi_ws281x/commit/62689c9bdd690adf340494cde8cc7401b859be60 ) but it still doesn't work.

Can you help me to make it work with my Pizero ? :)

tom-2015, 2017-04-12:
OK but first I have to buy a Zero W :)

Duwacell, 2017-04-14:
I can create a new SSH login for you on my zero W if you want to make some test without buying one

tom-2015, 2017-04-17:
It's OK, I bought one but will take a few days to deliver

Duwacell, 2017-04-17:
*-*

tom-2015, 2017-04-23:
I've added support for the RPi Zero W, also tested the code, fixed a bug and works with my Zero W.

Duwacell, 2017-05-06:
I found the time to try your upgraded script. It  works pretty well.

Many thanks!

tom-2015, 2017-05-07:
OK then we can close the issue :)

closed 2017-05-07 by tom-2015