tom-2015/rpi-ws2812-server #5: Connecting to server and executing php script

electronlove, 2017-08-29:
Is there any obvious reason that I would not be able to run the server on a Pi running stretch ?  The server end seems to hang after connecting a client with a series of unknown cmd's.  The web browser does nothing.

tom-2015, 2017-08-29:
I haven't tried on stretch yet but it's possible they changed some stuff breaking compatibility...
I'll try to check it next weekend.

electronlove, 2017-08-30:
Does the the server directory require to be setup in a specific directory, including the php script ?  I installed in the pi directory and ws2812svr works fine as a console.  How are you initializing the php script ?  I set a socket to a port, say 9999 and when I go to a browser  and type,  i.e. 127.0.0.1:8080/php/index.php nothing happens on the client side either on the Pi (through RDP) or on another desktop or phone.  --- So maybe I am running ws2812svr wrong ?  I did go and install php, apache, so forth to make sure php was running.

tom-2015, 2017-08-30:
You need to setup a webserver (apache) and PHP for this to work. see: https://www.raspberrypi.org/documentation/remote-access/web-server/apache.md
The PHP script needs to be in the /var/www/html directory (this will be created when you setup apache).

electronlove, 2017-08-30:
Thanks.  Working now.

closed 2017-08-31 by tom-2015