tom-2015/rpi-ws2812-server #7: Error set SO_KEEPALIVE

charlie-cheetah, 2018-03-06:
Hi,
I made a web page with a colorpicker that is sending a fill command to tcp socket immediately when color is changed. And I stumble upon this error, when I'm using the script:

Waiting for client to connect.
Error set SO_KEEPALIVE

The script is working fine for a while, and then this error occur. These two lines are displaying in a infinite loop. I think it may be because the script is sending new command too fast.

tom-2015, 2018-03-06:
The error is generated in the PHP code?
What does the PHP script look like?

charlie-cheetah, 2018-03-06:
It's a copy paste from example:

```
function send_to_leds ($data){  
   $sock = fsockopen("192.168.1.102", 9999);  
   fwrite($sock, $data);  
   fclose($sock);  
}
```

tom-2015, 2018-03-07:
maybe the scrip is sending too much too fast, try adding usleep(10000); after the fwrite()

tom-2015, 2020-07-04:
Should be fixed now

closed 2020-07-04 by tom-2015