tom-2015/rpi-ws2812-server #8: Receive HTTP/GET Request errors in CLI when trying to access Webserver

Orm1server, 2018-03-11:
Get the errors:

unknown cmd: GET
unknown cmd: Host:
unknown cmd: Connection:
unknown cmd: Upgrade-Insecure-Requests:
unknown cmd: User-Agent:
unknown cmd: Win64
unknown cmd: x64)
unknown cmd: Accept:
unknown cmd: q=0.9,image/webp,image/apng,/*/
unknown cmd: q=0.8
unknown cmd: DNT
unknown cmd: Accept-Encoding:
unknown cmd: Accept-Language:
unknown cmd: q=0.9

Have tried from mobile browser chrome and desktop browser Internet Explorer and Chrome

FROM CLI system works great, webserver i cant load color picker as I get the above errors in CLI

tom-2015, 2018-03-11:
You can't use this directly in a browser, the ws2812 server program doesn't speak http.
You need to install a web server application like apache and PHP. PHP will act like an interface between the browser and the ws2812 program.

You can find information on how to install the web server here: https://www.raspberrypi.org/documentation/remote-access/web-server/apache.md

Once you have the web server working you remove the /var/www/html/index.html file and copy the files from the php folder to /var/www/html. The php folder is located in the directory of the ws2812 source code.

Then all you need to do is run sudo ./ws2812svr -tcp 9999

Orm1server, 2018-03-12:
Hello Tom,

I installed the app chew and php mod per the link you sent and am able to view the webpage with color picker but am still getting those CLI errors mentioned above.  Any suggestions?

tom-2015, 2018-03-12:
What is the url you enter in your browser?
Did you change anything to the PHP code?
What is the command you entered to start the ws2812svr program?

closed 2018-04-17 by Orm1server