tom-2015/rpi-ws2812-server #9: Hotspot feature to access webpage

Orm1server, 2018-03-12:
Hello,

First off new to GitHub,  my question and I think it in wrong place, but any chance you can include a hotspot that broadcasts the color picker webpage?  This rpi3 is going into my truck and won't have access to WiFi unless I specifically go out of way to set up hotspot.  Like I said prolly wrong area to ask but I'd still like to ask.

Thanks,
Steve

tom-2015, 2018-03-12:
You can configure the hotspot manually.
This page explains how to set up a hotspot https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md

After that you should be able to access the color picker by entering the IP of you Raspberry in your browser.

Orm1server, 2018-03-12:
Ok will try that thank you. Also awesome job at prompt responses

Orm1server, 2018-04-02:
Found raspap to be best success

closed 2018-04-17 by Orm1server