tom-2015/rpi-ws2812-server #10: Can't get it to work

AdamEricson, 2018-03-24:
Hello, this project is what iv'e been searching for

I manged to load the page where I can choose the colors but i get :  <?$brightness?> 

and nothing happens..

any idea? 

and yes I checked and the led strip does work.

I'm using a Rpi B 3 

Thank you!!!

AdamEricson, 2018-03-24:
OK update I got it to work!!
thank you.
but can i add more animations? 
And how can i control the brightness?

Thank you

tom-2015, 2018-03-25:
Use the brightness command
brightness 
		<channel>,		 #channel number to change brightness (default 1)
		<brightness>,	 #brightness to set (0-255, default 255)
		<start>,		 #start at this led number (default 0)
		<len>

You can also adjust the RGB color value like: FF0000 is very bright red and 0F0000 will also be the same red but less bright.

AdamEricson, 2018-03-25:
I'll try that next week I'm at the army now..

How can i add animations to the web page? 

Thank you very much!

tom-2015, 2018-03-31:
That requires PHP programming and web stuff. It's not really something you can do in 5 minutes.

AdamEricson, 2018-03-31:
I see, hopefully when i get time I'll learn all of this stuff..

Thank you

AdamEricson, 2018-03-31:
How can i make the server program lunch automatically at boot?

tom-2015, 2018-04-22:
try using crontab:

crontab -e

if they ask which editor to use, choose nano
move down and write at the last line:

@reboot sudo /location_to_your_server_program_here

CTRL+X to close the editor and choose Y to overwrite the file.

AdamEricson, 2018-04-22:
Will do!

closed 2019-12-26 by tom-2015