tom-2015/rpi-ws2812-server #11: Load predefined script from web gui.

Orm1server, 2018-04-02:
Hey thanks for an awesome product. Got it working great with all static colors from webgui AP.  

Anyways I'm trying to control individual neopixels or run a script that controls individual neopixel from web gui.  Do you have any suggestions?

Orm1server, 2018-04-17:
Hey there. Been a couple weeks. Just poking the question hoping for an answer

maxiimilian, 2018-04-17:
Hey :) I have something similar in mind. Not individual pixels but ranges of pixels to group them into a "virtual" light. From a programmatically point, however, it does not matter if you want to talk to individual or ranges of pixels.
Do you have a certain language in mind?

Orm1server, 2018-04-17:
If you look at bottom of read me there is a way to load a file currently in
creators program with what youre talking about but I can't figure out how
to load the file from the web interface, I can only see instructions for
CLI. I'm not a dev but was hoping for this feature

maxiimilian, 2018-04-17:
No I am not the developer nor anyhow related to the project. I just came across it and use it for my own UI.. Are you running a php backend as well? If so, try to understand the exex_shell command. It allows you to run code in your cli triggered by a website. If not you can use plain js as well. You only need to modify the data your Javascript sends to the server. Those data can be the content of the file you created with your instructions. The great thing about this server is the standardised interface. It does not matter if you enter the commands I interactively in the shell, through a file or by socket through Javascript or PHP.

tom-2015, 2018-04-17:
In your PHP script try something like:

$commands_from_file=file_get_contents("/home/pi/commands.txt");
send_to_leds($commands_from_file);

replace the /home/pi/commands.txt with the location of the file containing the commands you want.
Don't forget the www user needs to have permission to read the file.

Orm1server, 2018-04-17:
Awesome thank you both of you. I will try that

closed 2020-07-04 by tom-2015