tom-2015/rpi-ws2812-server #17: Pass setup parameters on server start?

alphapapa77, 2018-08-23:
Any way to pass setup parameters on start?

Like:
`sudo ./ws2812svr -s 1,50 -tcp 9999`

Or is there a way to find out if the server has already been initialized?

tom-2015, 2018-09-02:
not yet but can be added I think

tom-2015, 2018-09-30:
This is now possible with -i command option.

sudo ./ws2812svr -i "setup 1,4,5;init;"

alphapapa77, 2018-10-01:
This is great news - thank you for creating this awesome program

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality6&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality6&>
10/01/18,
7:33:29 PM

closed 2020-07-04 by tom-2015