tom-2015/rpi-ws2812-server #20: Segmentation Fault on 3rd init over TCP connection

CampbellCrowley, 2018-11-14:
I'm using NodeJS v8.12.0 to open a TCP client socket using this script:
```
const net = require('net');

let client = new net.Socket();

client.connect(80, '192.168.1.211', () => {
  console.log('Connected!');

  client.write('setup 1,150;');
  client.write('init;');
  client.end();
  client.destroy();
});
```
After third execution of this the server crashes with a Segmentation Fault.

GDB output and stack trace:
```
pi@Pi-2:~/rpi-ws2812-server $ sudo gdb ws2812svr
GNU gdb (Raspbian 7.12-6) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.                       
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ws2812svr...done.
(gdb) run -tcp 80             
Starting program: /home/pi/rpi-ws2812-server/ws2812svr -tcp 80
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Listening on 80.              
Waiting for client to connect.
Client connected.             
Waiting for client to connect.
Client connected.             
Waiting for client to connect.
Client connected.             
                 
Program received signal SIGSEGV, Segmentation fault.
__GI___getpagesize () at ../sysdeps/unix/sysv/linux/getpagesize.c:28
28      ../sysdeps/unix/sysv/linux/getpagesize.c: No such file or directory.
(gdb) bt         
#0  __GI___getpagesize () at ../sysdeps/unix/sysv/linux/getpagesize.c:28
#1  0x000169fc in unmapmem (addr=0x76ff10a0, size=8) at mailbox.c:70
#2  0x00017c88 in unmap_registers (ws2811=0x2c3b0 <ledstring>) at ws2811.c:258
#3  0x00019900 in ws2811_fini (ws2811=0x2c3b0 <ledstring>) at ws2811.c:1078
#4  0x00011afc in init_channels (args=0x0) at main.c:283
#5  0x000159a8 in execute_command (command_line=0x2d828 "init") at main.c:1542
#6  0x00015d38 in process_character (c=59 ';') at main.c:1611
#7  0x00016688 in main (argc=3, argv=0x7efff794) at main.c:1780
```

Compiled without PNG or JPEG support using patch from #18

tom-2015, 2018-11-14:
It seems to crash when uninitializing a previous setup. Try to call setup and init only once at startup of the program. You can use the -i command line parameter for this.
sudo ./ws2812svr -i "setup 1,150;init;"

Then only use commands to change the leds in node red see if that solves the problem.

CampbellCrowley, 2018-11-14:
This does seem to solve the the segfaults, thanks. I was not aware that both the -i and -tcp parameters would work together.
```sudo ./ws2812svr -i "setup 1,150;init;" -tcp 80``` works great

However, am I right in assuming that I would not be able to change the number of LEDs that are connected without restarting the server each time they change?

This isn't a huge issue for me, but would mean that I need to update 2 devices (client and server) every time I change the number of LEDs that are connected, instead of just the client.

Thanks for making this awesome thing!

tom-2015, 2018-11-17:
Theoretically it should be possible to do multiple inits but seems like there is a bug in the mailbox.c but this is not my code. So will not be easy to fix I think...

tillig, 2019-01-23:
I tried to reproduce this because I was also noticing things hanging/crashing but haven't been able to repro it. I seem to be able to send the setup/init commands with every script/connection. However, I do seem to be having a different issue where the script _length_ may exceed some maximum, for which I'll add a separate issue.