tom-2015/rpi-ws2812-server #23: Raspberry Pi 3 freezes while using php

razorpl, 2019-01-02:
Im using apache+php on raspbian with Your index.php in /var/www/html/. 
When I run index.php and change color via Web-gui system freezes, after while and I'm loosing WLAN connection.
Im running server with: ```sudo ./ws2812svr -i "setup 1,200,1;init;" -tcp 9999```

I'm getting issues with running Your php code as well:
http://192.168.0.109/index.php?cmd=fill&color=CC0066&brightness=%3C?echo%20$brightness?%3E
And the text field always show: ```<?$brightness?>``` after every color change

tom-2015, 2019-01-02:
Try setting "short_open_tag = On" in the PHP ini configuration file. You lose WLAN connection on the pi?

razorpl, 2019-01-02:
I'll try tomorrow. Yes, it completely hangs wlan. No communication via ssh and ports 80 or 9999 However leds are still on. I've got similar situation using audio reactive led library, the program was running and the leds reacts to audio but wlan crashed after few minutes and the program was still on and working. Using other projects went without issues. Im curious if it can be setup itself. Im running ws2811grb strip using GPIO_18 (gpio_gen1) which has number 12 on datasheet (the upper one) on raspberry pi 3b

razorpl, 2019-01-03:
Ok, setting "short_open_tag = On" fixed problem with displaying index.php but i still have issue with freezing, and server sometimes return ```munmap error
: Invalid argument``` errors my setup is ```setup 1,200,1```

tom-2015, 2019-01-23:
Could be the same problem as #25