tom-2015/rpi-ws2812-server #26: loop speed slow via PHP not when just using a TCP Connection

simonwhill, 2019-04-27:
Hi,
When I run a loop using PHP i seem to hit a speed limit but yet the same loop sent via TCP alone it runs much faster

For example this loop 
`setup 1,113,3; init; rainbow;render;thread_start;do;do;rotate;render;delay 10;loop 250;do;rotate 1,1,2;render;delay 10;loop 250;loop;thread_stop;`  
Sent using putty in raw mode to the ws2812-server instance on port 9999 runs quickly.
But the same loop in a php page runs much slower.
`case 'random':
			      $data .= "rainbow;brightness 1,$brightness;render;
                      thread_start;
                 do;
                     do;
			rotate;
			render;
			delay 10;
			loop 250;
                     do;
                        rotate 1,1,2;
                        render;
                        delay 10;
                    loop 250;
                loop;
                  thread_stop;";`

any ideas would be greatly appreciated

Simon

tom-2015, 2019-05-01:
Maybe the php script is using too much CPU?
Try running the command "top" to see if there is a process using all CPU.

simonwhill, 2019-05-01:
Hi Tom,

Thanks for coming back to me, but no, its using around 5 - 10% of one core, so i'm not sure that's the case.

Any other thoughts / feedback greatly appreciated 

Simon

tom-2015, 2019-05-01:
Which IP does php script use? Only difference to me is that the php is connecting from the Raspberry itself and Putty does from an external machine. But don't have any idea what can be wrong at this moment.

simonwhill, 2019-05-01:
HI Tom,

Its connecting to localhost (127.0.0.1) and my script is connecting to its local IP address (10.99.32.112).

I have tried Apache and Nginx and have similar results.

Thanks

Simon

`<?php

$cmd = @$_REQUEST['cmd'];
$brightness = intval(@$_REQUEST['brightness']);
$color = @$_REQUEST['color'];
if ($color=='') $color='FF0000';
if ($brightness==0) $brightness=128;

$data  = "setup 1,28,3,0,255,18;setup 2,113,3,0,255,19;init;";

if ($cmd!=''){
    switch ($cmd){
        case 'fill':
            $data .= "fill 1,$color;fill 2,$color;render 1;render 2;";
            break;
        case 'rainbow':
            $data .= "brightness 1,$brightness;rainbow;render;";
            break;
		    case 'random':
			      $data .= "rainbow 1;rainbow 2;render 1;render 2;
                      thread_start;
                 do;
                     do;
                        rotate 1,1,0;
                        rotate 2,1,0;
                        render 1;
                        render 2;
                        delay 10;
                     loop 250;
                     do;
                        rotate 1,1,1;
                        rotate 2,1,1;
                        render 1;
                        render 2;
                        delay 10;
                    loop 250;
                loop;
                  thread_stop;";
        case 'rotating_rainbow':
            $data .= "rainbow;brightness 1,$brightness;render;
                      thread_start;
                     do;
                         rotate 1,1,2;
                        render;
                        delay 200;
                     loop;
                    thread_stop;";
            break;
    }
    send_to_leds($data);
}

function send_to_leds ($data){
    $sock = fsockopen("127.0.0.1", 9999);
    fwrite($sock, $data);
    fclose($sock);
}
?>
<html>
<head>
  <style>
  @import url('https://fonts.googleapis.com/css?family=Nunito:400,600,700');
  body {
    margin:0;
    background-color:#191919;
    min-height:100vh;
    display:flex;
    align-items:center;
    justify-content:center;
    font-family:Helvetica,Sans-serif;
    font-family: 'Nunito', sans-serif;
  }
  a {
    text-decoration:none;
    color:#FFF;
  }
  .rainbow-button {
    width:calc(7vw + 2px);
    height:calc(3vw + 2px);
    background-image: linear-gradient(90deg, #00C0FF 0%, #FFCF00 49%, #FC4F4F 80%, #00C0FF 100%);
    border-radius:3px;
    display:flex;
    align-items:center;
    justify-content:center;
    text-transform:uppercase;
    font-size:1vw;
    font-weight:bold;
  }
  .rainbow-button:after {
    content:attr(alt);
    width:6vw;
    height:2vw;
    background-color:#191919;
    display:flex;
    align-items:center;
    justify-content:center;
  }
  .rainbow-button:hover {
    animation:slidebg 2s linear infinite;
  }

  @keyframes slidebg {
    to {
      background-position:20vw;
    }
  }
  .btn {
    border: 3px solid black;
    background-color: Black;
    color: black;
    padding: 14px 28px;
    font-size: 16px;
    cursor: pointer;
    border-radius: 5px;

  }

  /* Green */
  .Red {
    border-color: #ff0000;
    color: red;
  }

  .Red:hover {
    background-color: #ff0000;
    color: Black;
  }

  /* Blue */
  .Orange {
    border-color: #ff3300;
    color: #ff3300
  }

  .Orange:hover {
    background: #ff3300;
    color: Black;
  }

  /* Orange */
  .Yellow {
    border-color: #ffff00;
    color: #ffff00;
  }

  .Yellow:hover {
    background: #ffff00;
    color: Black;
  }

  /* Red */
  .Green {
    border-color: #00ff00;
    color: #00ff00
  }

  .Green:hover {
    background: #00ff00;
    color: Black;
  }

  /* Gray */
  .Blue {
    border-color: #0000ff;
    color: 0000ff;
  }

  .Blue:hover {
    background: #0000ff;
    color: Black;
  }
  /* Orange */
  .Cyan {
    border-color: #00ffff;
    color: #00ffff;
  }

  .Cyan:hover {
    background: #00ffff;
    color: Black;
  }

  /* Red */
  .Purple {
    border-color: #ff00ff;
    color: #ff00ff
  }

  .Purple:hover {
    background: #ff00ff;
    color: Black;
  }

  /* Gray */
  .Pink {
    border-color: #cc0066;
    color: cc0066;
  }

  .Pink:hover {
    background: #cc0066;
    color: Black;
  }

}

  </style>
</head>
<body>
<button onclick="clickColor('#FF0000')"class="btn Red">Red</button>
<button onclick="clickColor('#FF3300')"class="btn Orange">Orange</button>
<button onclick="clickColor('#FFFF00')"class="btn Yellow">Yellow</button>
<button onclick="clickColor('#00FF00')"class="btn Green">Green</button>
<br/>
<button onclick="clickColor('#0000FF')"class="btn Blue">Blue</button>
<button onclick="clickColor('#00FFFF')"class="btn Cyan">Cyan</button>
<button onclick="clickColor('#FF00FF')"class="btn Purple">Purple</button>
<button onclick="clickColor('#CC0066')"class="btn Pink">Pink</button>
<br/>
<a href="index.php?cmd=random&brightness=<?echo $brightness?>" class="rainbow-button" alt="Rainbow"></a>
<br/>
<script type="text/javascript">

function mouseOutMap(){

}

function mouseOverColor(color){

}

function clickColor(color,x,y){
   window.location='index.php?cmd=fill&color='+color.substr(1)+'&brightness=<?echo $brightness?>';
}

function set_brightness(){
    window.location='index.php?cmd=fill&color=<?echo $color?>&brightness='+document.getElementById('txt_brightness').value;
}

</script>
</body>
</html>`

tom-2015, 2019-05-01:
What happens if you send the text directly on the Raspberry without PHP?
Create a text file with nano (script.txt) and paste the ws2812 script in there.
Install netcat (sudo apt-get install netcat)
Try these commands:

netcat 127.0.0.01 9999 < script.txt

netcat 10.99.32.112 9999 < script.txt

see if this works as expected.
What happens if you make a simple script with only 1 loop and 2 delays of 1000ms and make LEDs flash. Do they flash at 1 second interval?

simonwhill, 2019-05-02:
right that seems to behave in the same way as the TCP connection from another PC, the scripts runs using netcat as it did from the test pc.

Simon

simonwhill, 2019-05-02:
i may have found something, i just looked at the debug and saw the following

TCP Connection (putty raw)

pi@RasPi-Test:~/rpi-ws2812-server $ sudo ./ws2812svr -d -v -tcp 9999
Listening on 9999.
Waiting for client to connect.
Client connected.
Initialize channel 0,28,3,0,255,18
Initialize channel 1,113,3,0,255,19
Init ws2811 800000,5
Rainbow 0,1,0,255,0,28,19697712
Rainbow 1,1,0,255,0,113,19697712
Render 1
Render 2
Write to thread buffer: do
Write to thread buffer: do
Write to thread buffer: rotate 1,1,0
Write to thread buffer: rotate 2,1,0
Write to thread buffer: render 1
Write to thread buffer: render 2
Write to thread buffer: delay 10
Write to thread buffer: loop 250
Write to thread buffer: do
Write to thread buffer: rotate 1,1,2
Write to thread buffer: rotate 1,1,1
Write to thread buffer: render 1
Write to thread buffer: render 2
Write to thread buffer: delay 10
Write to thread buffer: loop 250
Write to thread buffer: loop
Thread stop.
Running thread.
Waiting for client to connect.
Enter thread 1,0,138.
do 2
do 5
Rotate 0 1 0 0
Rotate 1 1 0 0
Render 1
Render 2
loop 67
Rotate 0 1 0 0
Rotate 1 1 0 0
Render 1
Render 2
loop 67
Rotate 0 1 0 0
Rotate 1 1 0 0
Render 1
Render 2
loop 67
Rotate 0 1 0 0
Rotate 1 1 0 0
Render 1
Render 2
loop 67

and then this is the PHP version, i guess its a syntax issue?

TCP Connection via PHP Script

pi@RasPi-Test:~/rpi-ws2812-server $ sudo ./ws2812svr -d -v -tcp 9999
Listening on 9999.
Waiting for client to connect.
Client connected.
Initialize channel 0,28,3,0,255,18
Initialize channel 1,113,3,0,255,19
Init ws2811 800000,5
Rainbow 0,1,0,255,0,28,32759856
Changing brightness 0, 128, 0, 28
Render (null)
Write to thread buffer: do
Write to thread buffer: rotate 1,1,2
Write to thread buffer: render
Write to thread buffer: delay 200
Write to thread buffer: loop
Thread stop.
Running thread.
Waiting for client to connect.
Enter thread 1,0,38.
do 2
Rotate 0 1 2 0
Render (null)
loop 37
Rotate 0 1 2 0
Render (null)
loop 37
Rotate 0 1 2 0
Render (null)
loop 37
Rotate 0 1 2 0
Render (null)
loop 37
Rotate 0 1 2 0
Render (null)
loop 37
Rotate 0 1 2 0
Render (null)
loop 37

simonwhill, 2019-05-02:
OK i feel stupid!!! i found the fault

the switch statement was missing one of the break elements!

Thanks for your help Tom

Simon

closed 2019-05-02 by simonwhill

tom-2015, 2019-05-02:
haha yes I know that kind of mistake :)