tom-2015/rpi-ws2812-server #29: Program takes a long time to respond

davidpox, 2019-06-19:
Hi, I'm currently using this program on a RPI 3B+ using : `sudo ./ws2812svr -i "setup 1,60;init;" -tcp 9999` and I'm running the php script here: https://github.com/tom-2015/rpi-ws2812-server/blob/master/php/index.php on a remote machine on the local network (I changed the IP in fsockopen)

So far, only the rainbow/brightness commands work - however they take a long time to respond (a minute or two) and react. The rotating rainbow never responded even after 10 minutes. 
The program outputs 'Client Connected' as soon as I press an option on the website, and then after the colour changed after a few minutes, it says 'Waiting for client to connect'. I checked CPU usage with `top` and it remains at around 25% while executing, and 1.3% when done.

I've tried the following #28 and that did not succeed.

davidpox, 2019-06-19:
Turns out I was using the wrong LED Type. Setting it to 2 (WS2811_STRIP_GRB) gives correct colours, and response time is much faster, but there's still some severe delays and the rotating rainbow doesn't work, still.

tom-2015, 2019-06-20:
Can you post the PHP code you are using?

davidpox, 2019-06-20:
```php
<?php
$cmd = @$_REQUEST['cmd'];
$brightness = intval(@$_REQUEST['brightness']);
$color = @$_REQUEST['color'];
if ($color=='') $color='FF0000';
if ($brightness==0) $brightness=32;
$data  = "";
if ($cmd!=''){
    switch ($cmd){
        case 'fill':
            $data .= "brightness 1,$brightness;fill 1,$color;render;";
            break;
        case 'rainbow':
            $data .= "brightness 1,$brightness;rainbow;render;";
            break;
        case 'rotating_rainbow':
            $data .= "brightness 1,$brightness;
                      thread_start;
                     do;
                        rotate 1,1,2;
                        render;
                        delay 200;
                     loop;
                    thread_stop;";
            break;
    }
    send_to_leds($data);
}
function send_to_leds ($data){
    $sock = fsockopen("192.168.XXX.XXX", 9999);
    fwrite($sock, $data);
    fclose($sock);
}
?>
```

tom-2015, 2019-06-24:
Can't see problem, what happens if you run the same commands from a text file or direct enter on CLI?
Try connecting to the server with putty (raw mode) and paste the code to see if it's PHP related or not.