tom-2015/rpi-ws2812-server #33: ws2811_init failed with code -3 (Hardware revision is not supported)

raquintanilla, 2019-12-12:
Any suggestion on a RPI 4

tom-2015, 2019-12-12:
Strange it is supposed to work... Or did they make a new revision?

tom-2015, 2019-12-12:
Ok found there is a V1.2 revision of Rpi 4, can you pull the changes and try recompile?

raquintanilla, 2019-12-19:
I tried it with no success, same error

tom-2015, 2019-12-19:
Did you download and recompile the changes in rpihw.c ?
Can you check your Raspberry HW revision with command:
`cat /proc/cpuinfo`

raquintanilla, 2019-12-19:
It looks like there are no V1.2 (Fake news)

frid1993, 2020-12-11:
ws2811_init failed with code -3 (Hardware revision is not supported) #33

pi@raspberrypi:~ $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2711
Revision        : c03114
Serial          : 10000000458935bf
Model           : Raspberry Pi 4 Model B Rev 1.4

tom-2015, 2020-12-12:
I've added the revisoon to the rpihw.c, can you try and see if it works.
Strange the revision number is not in the official documentation.
[https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md](url)

frid1993, 2020-12-17:
Thank Tom. Seems to be working now!

Spangen, 2022-02-23:
Seems to be several revisions of compute module 4.
I have this one that gives         .hwver = 0xc03140,

[Raspberry Pi Compute Module 4]
Wireless: yes
Ram: 4 GB
eMMC storage: 32 GB
________________________________________________________________
pi@Pi-Noddy:~ $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 126.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 126.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 126.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 126.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2711
Revision        : c03140
Serial          : 10000000f7d987f1
Model           : Raspberry Pi Compute Module 4 Rev 1.0

tom-2015, 2022-02-24:
Ok I'll have to add this to the rpihw.c file