tom-2015/rpi-ws2812-server #34: Issue with -p flag. The animation is not smooth

leoheck, 2019-12-14:
Hi, I've noticed that running the same animation through "-p" and `-f` yields different behaviors.

I made a smooth animation using a sile and the `-f` flag. 

```
setup 1,8,3
init

fill 1,ff3300,0,8

fade 1,0,80,16
render

delay 100

do
    fade 1,79,10,16
    fade 1,11,80,26
    delay 100
    render
loop 3

fade 1,80,0,16
```

Now, converting it to a script using the `-p` is not smooth anymore,

Can you confirm?

```bash

led_channel=1
led_count=8
led_type=3
led_invert=0
led_gpio=18
led_brightness=20

rgb_yellow=ffb100

read -r -d '' animation <<- EOM
setup ${led_channel},${led_count},${led_type},${led_invert},${led_brightness},${led_gpio}
init

fill 1,${rgb_yellow},0,8

fade 1,0,80,16
render

delay 100

fade 1,79,10,16
fade 1,11,80,26
delay 100
render

fade 1,79,10,16
fade 1,11,80,26
delay 100
render

fade 1,79,10,16
fade 1,11,80,26
delay 100
render

fade 1,80,0,16
EOM

echo "$animation" > /dev/ws281x
```

It looks like the fade command is different between these two versions

leoheck, 2019-12-14:
I just removed the last part of my command and it looks like it working better

Replacing this

```
setup ${led_channel},${led_count},${led_type},${led_invert},${led_brightness},${led_gpio}
```

with this

```
setup ${led_channel},${led_count},${led_type}
```

tom-2015, 2019-12-16:
I was not at home this weekend will take a look at it later

leoheck, 2019-12-16:
Sure, when you have time.