tom-2015/rpi-ws2812-server #36: How to use the `-tcp` mode?

leoheck, 2019-12-18:
First of all, the README.md shows
[image]

The command has the port missing, it should be like
```
sudo ./ws2812svr -tcp [PORT]
```

Also 
How could I use the `-tcp` mode with the command line.

I am trying this:
```
# Launch the server
sudo ./ws2812svr -tcp 12346

# Send some commands
echo "setup 0,8,3; init; fill 1,FFFFFF; render" | nc  127.0.0.1 12346
``` 
It gives me
```
Client connected.
Invalid channel number, use channels <number> to initialize total channels you want to use.
Gpio 0 is illegal for LED channel 0
ws2811_init failed: Selected GPIO not possible
Invalid channel number, did you call setup and init?
```

And the server closes, which is not good because someone can send some wrong command to just close the server. 

Trying to send the same command from the README, like
```
echo "setup channel_1_count=8;rainbow;brightness 1,100;" | nc -k 127.0.0.1 9999
```

[image]

leoheck, 2019-12-18:
Actually I made it work with this command (the README.md lacks the `init` and the `render` commands.

```echo "setup 1,8,3;init;rainbow;brightness 1,100;render;" | nc 127.0.0.1 9998```

leoheck, 2019-12-18:
Here, the example for a loop, that disconnects the client after sending the command.
It is working

```bash
echo "$cmd" | nc -w 0 127.0.0.1 9997

read -r -d '' cmd <<-EOM
setup 1,8,3
init
rainbow
thread_start
do
	rotate 1,1,2
	render
	delay 200
loop
thread_stop
EOM

echo -e "$cmd" | nc 127.0.0.1 9998
```

leoheck, 2019-12-18:
TCP version works better than using file or uname pipe. Thanks for referring me that.

tom-2015, 2019-12-19:
you're right the documentation was not up to date. I've updated that now also added 2 new effects: chaser and color_change.

leoheck, 2019-12-19:
This is still wrong
```sudo ./ws2812svr -tcp```

It should be like this unless you added a default port in the code.
```sudo ./ws2812svr -tcp [PORT]```

tom-2015, 2019-12-24:
yes default port is 9999