tom-2015/rpi-ws2812-server #37: render does weird things

hans-martijn, 2020-01-16:
I have a test webpage to send a render to my server (render 1,0,FF0000FF0000FF0000;), the first light turns red, the second and the third turn blue. If I render with 00FF0000FF0000FF00, the first one is green, the second one is red, the third one is off.

Only if I put two extra hex numbers in between the colour-codes it works. I initialized them with type 4, which should be right. I checked the chip (and even the manufacturer's site; it is a 24 bit chip. The two extra hex numbers don't change anything on the chip either. Do I initialize wrong? (setup 1,3,4;init).

tom-2015, 2020-01-16:
what happens when you do a fill?

fill 1,FF0000;
render;

hans-martijn, 2020-01-16:
That works, they all become red. The reason I want to use render is that I want to be able to control each light separately.

tom-2015, 2020-01-16:
It is a bug in the setup command, I've updated try pull & make see if this fixes the problem

hans-martijn, 2020-01-16:
Yes, it works like a charm!

Thanks a lot!!!

Regards,
Hans

closed 2020-01-20 by hans-martijn