tom-2015/rpi-ws2812-server #40: Two LED strip on RPI 4

MdotWym, 2020-02-03:
Hi

Thank you for your WS2812 server.

I tried your it with two LED strips. It worked from the beginning very nice with only one strip.
Then I connected a second, which is not working.
For the first, the setup is:
`setup 1,256,3,0,255,18;
init;
` 

Using the second, the setup is:
`setup 1,256,3,0,255,18;
setup 2,60,3,0,255,13;
init;`

The first one uses PWM0, the second PWM1 (I guess). So in theory two strips can be connected.
Before going into details I'd like to know if I'm doing something fundamentally wrong.

Do you prefer another channel to ask such basic questions?

BR
Wym

MdotWym, 2020-02-03:
Update:
I'm really sorry for my request.
It works fine if using the right GPIO on the board!
After checking all possible root causes, finaly found the bug.
--> Issue closed.

BR
Wym

MdotWym, 2020-02-04:
Hi
Basically the problem (wiring) is solved. After some tests and reading the documentation the following problem has arisen:

`thread_start
   do  
      rotate 1,1,2  
     render  
     delay 200  
  loop  
thread_stop`

MdotWym, 2020-02-04:
Hi
Basically the problem (wiring) is solved. After some tests and reading the documentation the following problem has arisen:

As it seems it is not possible to operate both LED bands at the same time. If a command is sent via TCP for the first strip, and shortly afterwards a command for the second strip, both processes are aborted.
After the first command is completed, the TCP connection is closed. As soon as the second TCP command opens again, the first process is stopped. Interestingly, the second process is also stopped. This is partly in accordance with the note in the documentation under the topic "thread".

Is it possible to run two strips independently of each other?
Could theoretically the server be started twice? One for Stripe 1 and the other for Stripe 2?

BR
Wym

tom-2015, 2020-02-04:
Sorry, not possible to do this unless you can combine commands for strip 1 and 2 in the same loop/thread.

Try running 2 servers on different ports.

MdotWym, 2020-02-05:
Ok, tried running with two servers. In summary, the following can be said.

The first instance of the server is set up and initialized on channel 1 with PWM0 with the pin used for LED strip 1. TCP port 9999.

The second instance of the server is set and initialized on channel 1 with PWM0 and a UNUSED pin and on **channel 2** with **PWM1** and the pin used for LED strip 2. TCP port 9998.

Channel 1 cannot be used together with PWM1 since an error during setup is thrown by the server. So I had to setup an "unused" channel 1 with an unused pin on this second server.

If messages are now sent one after the other, Strip 1 and Strip 2 will work properly. However, if the processes run parallel, the LEDs of both strips light up, but they influence each other. So it does not work just like that.

Could this be because both servers or the rpi_ws281x library use the same DMA channel? Maybe you have to assign different DMA channels to each server?

Thank you for your support.

MdotWym, 2020-02-05:
Update: 
If both servers are setup with two different DMA channels, nothing works anymore.
Both channels are dead. For the first server DMA 5 is used, for the second server DMA 4.
I tried DMA 8 as well, but it doesn't work either.

tom-2015, 2020-02-08:
You forgot the first argument of the init command is the operating frequency.
try:
init 800000,10;

but I've changed the default DMA number to 10 anyway in the source code.