tom-2015/rpi-ws2812-server #41: Startup Issue after Update to new Version

Bigdeath, 2020-02-03:
After i updated the rpi-ws2812-server my startup script no longer works as before.

Here my Script:

[Unit]
Description=LED Server
Wants=network-online.target
After=network-online.target
StartLimitIntervalSec=0

[Service]
Type=forking
Restart=always
RestartSec=1
User=pi
WorkingDirectory=/home/pi/rpi-ws2812-server
ExecStartPre=/bin/echo "LED-Server versucht zu starten"
ExecStart=/usr/bin/screen -dmS LEDSERVER bash -c "cd /home/pi/rpi-ws2812-server ; sudo ./ws2812svr -i 'setup 1,13,3;init;blink 1,FF0000,00FF00,75,24;fade 1,255,0,10,2;render;fill 1,000000;render;brightness 1,255;' -tcp 9999; bash" &
ExecStartPost=/bin/echo "LED-Server gestartet"
[Install]
WantedBy=multi-user.target
-------------------------

When the pi starts he showed me through this script that the ledserver is running. But now server starts without flashing my leds. When i send a TCP paket then it runs the blink.... and after that the TCP paket. I dont understand why. Server runs in a screen to debug for problems.

Second Problem: I tried it with your new startupscript, but it just keeps coming that i didnt run the init. No matter what I configure.

Everything worked wonderfully before the update.

I use your server to let viewers interact with my LEDs via twitch. Chats, follow = blink blink :-)
Nice gimmick! Thanks for your work!
Sorry for my bad english, im from Germany.

tom-2015, 2020-02-04:
The new startup script you mean running it as a service/with config file?
Did you enter the command to initialize in the config file?
Like in config file:
mode=tcp
port=9999
file=/home/pi/test.txt
pipe=/dev/leds
init=setup 1,13,3;init;blink 1,FF0000,00FF00,75,24;fade 1,255,0,10,2;render;fill 1,000000;render;brightness 1,255;

What happens if you run the command:
sudo ./ws2812svr -i 'setup 1,13,3;init;blink 1,FF0000,00FF00,75,24;fade 1,255,0,10,2;render;fill 1,000000;render;brightness 1,255;'

Bigdeath, 2020-02-04:
The new startup script you mean running it as a service/with config file?
**Yes**
Did you enter the command to initialize in the config file?
**Yes**
Like in config file:
mode=tcp
port=9999
file=/home/pi/test.txt
pipe=/dev/leds
init=setup 1,13,3;init;blink 1,FF0000,00FF00,75,24;fade 1,255,0,10,2;render;fill 1,000000;render;brightness 1,255;
**Yes**
What happens if you run the command:
sudo ./ws2812svr -i 'setup 1,13,3;init;blink 1,FF0000,00FF00,75,24;fade 1,255,0,10,2;render;fill 1,000000;render;brightness 1,255;'
**Then starts right. When i now put -tcp 9999 at the end, then the server starts without my start blinking and i have the same problem.**

Bigdeath, 2020-02-04:
sudo ./ws2812svr -i 'setup 1,13,3;init;blink 1,FF0000,00FF00,75,24;fade 1,255,0,10,2;render;fill 1,000000;render;brightness 1,255;'
-> Starts with the blink

sudo ./ws2812svr -i 'setup 1,13,3;init;blink 1,FF0000,00FF00,75,24;fade 1,255,0,10,2;render;fill 1,000000;render;brightness 1,255;' -tcp 9999
Terminal:
Listening on 9999.
Waiting for client to connect.
-> Starts without the Blink
Send TCP Paket:
Client connected.
-> Start Blink
Waiting for client to connect.
-> TCP Paket perform

I dont understand why the server perform the -i after a TCP paket.

Bigdeath, 2020-02-08:
i think i have found the problem, but have no solution.

When i start the server without TCP option, i can tell the server what i want inside shell.
But when i start the server with TCP option, shell commands doesnt work.
I think thats the point why my Service start script not work.

TCP and no shell command works:
****:/home/pi/rpi-ws2812-server# sudo ./ws2812svr -tcp 9999
Listening on 9999.
Waiting for client to connect.
setup test test
init
render

Without TCP, shell works
****:/home/pi/rpi-ws2812-server# sudo ./ws2812svr
setup test test
Invalid channel number, use channels <number> to initialize total channels you want to use.
init
Gpio 0 is illegal for LED channel 0
ws2811_init failed: Selected GPIO not possible
render
Invalid channel number, did you call setup and init?

instant info about my false command without tcp option. Now why?

tom-2015, 2020-02-08:
Can you check if the issue is gone with the latest version.

Bigdeath, 2020-02-09:
Jeah! Startup blink is now functional. But the Problem with TCP Option and no shell commands still there.

tom-2015, 2020-02-10:
If you chose TCP you cannot enter commands it's normal.

Bigdeath, 2020-02-10:
Ouh ok, I thought I could enter it despite TCP mode. Well, probably wrongly thought. Thanks anyway for the quick fix!

closed 2020-02-11 by tom-2015