tom-2015/rpi-ws2812-server #42: New command store/laststate possible?

Bigdeath, 2020-02-10:
Is it possible to make a store and laststate command?
example:
I am making a fill with red and will send a blink at some point. And say a laststate command within the blink command that it automatically returns to the last state in the case just red. You could give the fill example a store that the laststate knows what it has to do.

With the store command you could generate a file or save it at the memory that stores the whole command and calls it with laststate. Each store then overwrites the last store.

I hope you understand what I mean.

Greetings

Bigdeath, 2020-04-05:
No Answer about 1,5 months?

tom-2015, 2020-04-05:
Sorry, I've been busy with other project did not have time to implement yet.

Bigdeath, 2020-04-06:
ok, no problem. Thanks for the Answer.

tom-2015, 2020-07-03:
Added these commands (load_state and save_state) and now updated the documentation. Can you check if it is working properly?

Bigdeath, 2020-07-07:
I have now tried to test it with a few scripts. I just don't quite understand how it works. No CSV is created. I think I'm just doing it wrong. Where exactly should it be inserted? At the end? Is only the last status saved in itself or a whole effect?

In my case i would like to save the endles loop from this Script:

```
<?php
send_to_leds("
thread_start
brightness 2,255
blink 2,FFFFFF,000000,100,10
render 2
rainbow 2
render 2
do
       rotate 2
       render 2
       delay 1000
loop
brightness 2,255
fill 2, 000000
render 2
thread_stop
");
function send_to_leds ($data){
   $sock = fsockopen("192.168.178.29", 9997);
   fwrite($sock, $data);
   fclose($sock);
}
$timestamp = time();
$datum = date("d.m.Y - H:i:s", $timestamp);
echo $datum;
$datei = fopen("./triggers.txt", "a+");
$date=date("d.m.Y - H:i:s");
fwrite($datei, $daten."LEDPION: $date\r\n");
fclose($datei);
?>
```

tom-2015, 2020-07-07:
It saves the current led state (color and brightness).
So normally you put the 

`save_state 1, "/dev/shm/test.csv"`

At the end to save the current led state.