tom-2015/rpi-ws2812-server #43: Same TCP Paket not perform multiple times at some time

Bigdeath, 2020-04-06:
After a certain point in time a TCP packet is no longer executed if I send it twice. (same packet, different pakets work)

Server log shows "Client connected. Waiting for client to connect." any time. But only First and Third .. Fith Paket perform. Just restarting server every time helps. Then same paket perform multiple times.

It does not matter whether in a row or with a certain time interval. 

One of my scripts:

```
<?php
send_to_leds("
thread_start
fill 1,000000
brightness 1,155
render
blink 1,FFFFF,000000,100,3
render
fill 1,000000
brightness 1,255
render
thread_stop
");
function send_to_leds ($data){
   $sock = fsockopen("192.168.178.29", 9999);
   fwrite($sock, $data);
   fclose($sock);
}
?>
```
Other:

```
<?php
send_to_leds("
thread_start
fill 1,000000
brightness 1,255
render
rainbow
render
do
       rotate
       render
       delay 100
loop 30
fill 1,000000
brightness 1,255
render
thread_stop
");
function send_to_leds ($data){
   $sock = fsockopen("192.168.178.29", 9999);
   fwrite($sock, $data);
   fclose($sock);
}
?>

```

It doesn't matter which script. At some point it no longer works.

With larger scripts I sometimes get this error message after sending it a second time.: Warning max nested loops reached!
I understand this message but it makes no sense when nothing perform.

Other Problem:
Is there a trick how TCP packets are executed in sequence and not a packet that cancels the other?
The following problem: If I run a long script and then a second script is executed, it often happens that the long script should end a certain sequence but is stopped by the second script and the second script is immediately executed. I would like to prevent that. Scripts should always be executed completely before another script is executed.

I Hope you understand my problems, used Google Translator. My english is not the best.

Bigdeath, 2020-06-10:
No solution for at least 2 months or at least one answer?

Bigdeath, 2020-06-13:
Thx for your Answer. I dont know how to recompile. Im not a dev. And dont know which file. Then im waiting for a Update.

tom-2015, 2020-06-14:
Update the code, can you download it and compile/reinstall and check if problem is gone?

Bigdeath, 2020-06-14:
Have installed it. Give you Feedback after my next 2-3 Twitch Streams, should there be problems.

Could it be a problem that I have two servers running on one pi3b+? Sometimes it happens that server2 suddenly plays on server1. Server1 restart and the effect continues with server2 and server1 can be addressed again. If this happens, packets that server1 should play will play on server2.

 Thank you for your work in advance.

tom-2015, 2020-06-14:
do you run the servers manual from command line or as daemon?

Bigdeath, 2020-06-14:
As a Service. 2 Folders. And with Screen to to check if errors occur.

Bigdeath, 2020-06-15:
Ok, now i have the same problem again. 

Every time i send:
Client connected.
Waiting for client to connect.

But only first, third, fifth .. perform. 

Any Problem with my Service script?

`[Unit]
Description=LED Server
Wants=network-online.target
After=network-online.target
StartLimitIntervalSec=0

[Service]
Type=forking
Restart=always
RestartSec=1
User=root
WorkingDirectory=/home/pi/rpi-ws2812-server
ExecStartPre=/bin/echo "LED-Server versucht zu starten"
ExecStart=/usr/bin/screen -dmS LEDSERVER bash -c "cd /home/pi/rpi-ws2812-server ; sudo ./ws2812svr -i 'setup 1,13,3;init;blink 1,FF0000,00FF00,75,24;fade 1,255,0,10,2;render;fill 1,000000;render;brightness 1,255;' -tcp 9999; bash" &
ExecStartPost=/bin/echo "LED-Server gestartet"
[Install]
WantedBy=multi-user.target
`

tom-2015, 2020-06-15:
Can you start with -d parameter, this will send some debug info about commands that are received and what it's doing. Then post the output.

Bigdeath, 2020-06-17:
Now i have a small Debug:

Write to thread buffer: brightness 1,255
Write to thread buffer: render
Thread stop.
Running thread.
Waiting for client to connect.
Enter thread 1,0,110.
fill 0,0,0,13,0
Changing brightness 0, 155, 0, 13
Render (null)
blink 0, 15794175, 0, 100, 3, 0, 13
Render (null)
fill 0,0,0,13,0
Changing brightness 0, 255, 0, 13
Render (null)
Exit thread.
Client connected.
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,155
Write to thread buffer: render
Write to thread buffer: blink 1,FFFFF,000000,100,3
Write to thread buffer: render
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,255
Write to thread buffer: render
Thread stop.
Running thread.
Waiting for client to connect.
<<<<<<<<<<<<<<<<Send but nothing performs on my LEDs
Client connected.
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,155
Write to thread buffer: render
Write to thread buffer: blink 1,FFFFF,000000,100,3
Write to thread buffer: render
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,255
Write to thread buffer: render
Thread stop.
Running thread.
Waiting for client to connect.
Enter thread 1,0,110.
fill 0,0,0,13,0
Changing brightness 0, 155, 0, 13
Render (null)
blink 0, 15794175, 0, 100, 3, 0, 13
Render (null)
fill 0,0,0,13,0
Changing brightness 0, 255, 0, 13
Render (null)
Exit thread.
<<<<<<<<<<<<<<<<Send second time and it performs on my LEDs
Client connected.
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,155
Write to thread buffer: render
Write to thread buffer: blink 1,FFFFF,000000,100,3
Write to thread buffer: render
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,255
Write to thread buffer: render
Thread stop.
Running thread.
Waiting for client to connect.

Bigdeath, 2020-06-18:
i will give it a try. thx!

Bigdeath, 2020-06-18:
I have this error:

root@BigPi2:/home/pi/rpi-ws2812-server# make
gcc -g -I/usr/include -DUSE_PNG -DUSE_JPEG -c main.c -o main.o
main.c: In function ‘tcp_wait_connection’:
main.c:2078:13: error: too many arguments to function ‘perror’
   if (s!=0) perror("Error creating new thread: %d", s);
             ^~~~~~
In file included from main.c:2:0:
/usr/include/stdio.h:848:13: note: declared here
 extern void perror (const char *__s);
             ^~~~~~
makefile:41: die Regel für Ziel „main.o“ scheiterte
make: *** [main.o] Fehler 1

Bigdeath, 2020-06-22:
Client connected.
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,155
Write to thread buffer: render
Write to thread buffer: blink 1,FFFFF,000000,100,3
Write to thread buffer: render
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,255
Write to thread buffer: render
Thread stop.
Running thread.
Waiting for client to connect.
Enter thread 1,0,110.
fill 0,0,0,13,0
Changing brightness 0, 155, 0, 13
Render (null)
blink 0, 15794175, 0, 100, 3, 0, 13
Render (null)
fill 0,0,0,13,0
Changing brightness 0, 255, 0, 13
Render (null)
Exit thread.
Client connected.
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,155
Write to thread buffer: render
Write to thread buffer: blink 1,FFFFF,000000,100,3
Write to thread buffer: render
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,255
Write to thread buffer: render
Thread stop.
Running thread.
Error creating new thread: 11Waiting for client to connect.
Client connected.
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,155
Write to thread buffer: render
Write to thread buffer: blink 1,FFFFF,000000,100,3
Write to thread buffer: render
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,255
Write to thread buffer: render
Thread stop.
Running thread.
Waiting for client to connect.
Enter thread 1,0,110.
fill 0,0,0,13,0
Changing brightness 0, 155, 0, 13
Render (null)
blink 0, 15794175, 0, 100, 3, 0, 13
Render (null)
fill 0,0,0,13,0
Changing brightness 0, 255, 0, 13
Render (null)
Exit thread.

tom-2015, 2020-06-23:
Does this happen the second time you connect?
Because "Error creating new thread: 11" means out of resources usually because a too high number of threads were created. But this should not be possible unless maybe after a few 1000 connects/disconnects...

Bigdeath, 2020-06-23:
Yes every Second command. I Have max 3 connects every Minute (Blink for Twitch Chat message every 20 Seconds, if someone should write ) + Follow,Sub,Raid,Host,Donate Events + Own Chattrigger. But i'm a little Streamer with max 10-20 Viewers.  So not that much actually happens. I Think i have in my stream time from 8 pm to midnight ~150 LED trigger.

Can you adress every Thread a Number, to see how much triggers i have since the server starts? Simple Counting i mean.

On my Second RasPi for my Door, i have the same problem after ~3 months. That only triggers LEDs when the Door opened. This Happens max ~6-8 times a day.

Can it happen that there is a TCP overflow? Then all my Triggers are over TCP connections.

tom-2015, 2020-06-26:
Do you have enough free memory left? Maybe something else is using all memory on the pi?

Bigdeath, 2020-06-26:
Yes, only your LED Server is Running. 107MB from 926MB in Use. ~2% CPU. Swap Empty.

tom-2015, 2020-06-27:
I'm afraid I'm not able to reproduce your problem :(
Did a few 100 connects/disconnects/send script all seems to work fine.
On "raspberrypi3 4.19.57-v7+ #1244 SMP Thu Jul 4 18:45:25 BST 2019 armv7l".
What is your Raspbian version / which Raspberry Pi are you using?
Did the problem start after an update?
Can you try with a new SD card and fresh install?

Bigdeath, 2020-06-27:
Sounds not good :-(

First Pi:
Linux BigPi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l
Installed: 8. April 2019
PiHole (lighttpd), unbound, led server (Door led)
57.5 °C ~ 134 °F

Second Pi:
Linux BigPi2 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l
Installed: 1. Jan 2019
Apache2, led server (for the Twitch things)
46,7 °C ~ 115 °F

Both 3B+
Different SD Cards and different ages.
Both PHP 7.0.33-0+deb9u7 (cli) (built: Feb 16 2020 15:11:40) ( NTS )
Cooled with Fan and heatsinks

Problem since i use it. :-(
If I send something to the LED server like a crazy one, I can't get it to have this error.
Like a DOS attack, have a HTML/JS site with buttons and send much different trigger. No Problem all works fine. But 2-3 hours livestream and then i have the problem again. I dont understand it. Twitch trigger do the same i do with my HTML/JS site: Fetch a php file.

Unfortunately I have no SD card left to test it currently. Have done a Benchmark for both SD Cards. All fine.

tom-2015, 2020-06-28:
Is the ws2812 server TCP port (9999) reachable from the public internet or only local LAN?
Maybe put a log in the PHP see how many requests you get and forward to the ws2812.

The livestream is it generated by the same Pi as the LEDS?
Maybe let the Pi run for a few hours with only the ws2812 server disconnected from public internet to see if any other software is causing the problem.

Bigdeath, 2020-06-28:
No for Security Reasons the Pi not reachable over the Puplic internet. 

It's a bit complicated, but I'm trying to explain it. A command that is written triggerd a so-called chat bot. This chat bot turns a fetch to a php file that resides on a server here at home. This php file shows which TCP packet should be sent to my RasPi in order to carry out the LED effects. So the actual TCP packet is only transmitted locally at home. Not on the internet. And that also works perfectly. 

Short: Twitch -> Chatbot -> Server at home -> RasPi 

And this works without problems. 

I try to put in all files a trigger to a text file with timestamp and triggername. Then i have a Overview how much triggers come to my server.

I don't understand the point with the other software. If I just run the server and e.g. 3 days not in use and then sending several triggers, it works. So it has nothing to do with the runtime of the RasPi or the ws2812 server. I only have the problem if the server really has to do something again and again.

Thank you very much for continuing to try to locate the problem. I and my viewers like this self-made LED brand and I would like to keep it running without having to constantly restart the server.

Bigdeath, 2020-06-28:
Now after my 5h Stream:
313 Chat LED Trigger
2 Other Custom LED Trigger

And i have the Problem again.

Bigdeath, 2020-06-29:
Here my Scripts:

Chat:

<?php
send_to_leds("
thread_start
fill 1,000000
brightness 1,155
render
blink 1,FFFFF,000000,100,3
render
fill 1,000000
brightness 1,255
render
thread_stop
");
function send_to_leds ($data){
   $sock = fsockopen("192.168.178.29", 9999);
   fwrite($sock, $data);
   fclose($sock);
}
$timestamp = time();
$datum = date("d.m.Y - H:i:s", $timestamp);
echo $datum;
$datei = fopen("./triggers.txt", "a+");
$date=date("d.m.Y - H:i:s");
fwrite($datei, "Chat: $date\r\n");
fclose($datei);
?>

And the two others:

<?php
send_to_leds("
thread_start
fill 1,000000
render
fill 1,00FFFF,0,1
brightness 1,255
render
delay 50
render
do
do
rotate 1,1,0
render
delay 100
loop 12
do
rotate 1,1,1
render
delay 100
loop 12
loop 3
render
blink 1,00FFFF,880000,100,10
render
fade 1,255,0,5,1
render
fill 1,000000
brightness 1,255
render
thread_stop
");
function send_to_leds ($data){
   $sock = fsockopen("192.168.178.29", 9999);
   fwrite($sock, $data);
   fclose($sock);
}
$timestamp = time();
$datum = date("d.m.Y - H:i:s", $timestamp);
echo $datum;
$datei = fopen("./triggers.txt", "a+");
$date=date("d.m.Y - H:i:s");
fwrite($datei, "Follow: $date\r\n");
fclose($datei);
?>

---------------------------------------------

<?php
send_to_leds("
thread_start
fill 1,000000
render
do
fill 1,4B0082,0,1
fill 1,4B0082,12,1
render
delay 50
fill 1,000000
render
fill 1,0000FF,1,1
fill 1,0000FF,11,1
render
delay 50
fill 1,000000
render
fill 1,00FF00,2,1
fill 1,00FF00,10,1
render
delay 50
fill 1,000000
render
fill 1,FFFF00,3,1
fill 1,FFFF00,9,1
render
delay 50
fill 1,000000
render
fill 1,FF7F00,4,1
fill 1,FF7F00,8,1
render
delay 50
fill 1,000000
render
fill 1,FF4500,5,1
fill 1,FF4500,7,1
render
delay 50
fill 1,000000
render
do
fill 1,FF0000,6,1
render
delay 75
fill 1,000000
render
delay 75
loop 2
render
loop 10
delay 100
fill 1,000000
brightness 1,255
render
thread_stop
");
function send_to_leds ($data){
   $sock = fsockopen("192.168.178.29", 9999);
   fwrite($sock, $data);
   fclose($sock);
}
$timestamp = time();
$datum = date("d.m.Y - H:i:s", $timestamp);
echo $datum;
$datei = fopen("./triggers.txt", "a+");
$date=date("d.m.Y - H:i:s");
fwrite($datei, "Guenny: $date\r\n");
fclose($datei);
?>

Any Problems with this scripts?

tom-2015, 2020-06-30:
Scripts are OK don't know what can be wrong a.t.m.
I've added code to generate error text when pthread_join() fails, can you try compile and run current version?

Bigdeath, 2020-06-30:
Only have this:

Client connected.
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,155
Write to thread buffer: render
Write to thread buffer: blink 1,FFFFF,000000,100,3
Write to thread buffer: render
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,255
Write to thread buffer: render
Thread stop.
Running thread.
Error creating new thread: 11Waiting for client to connect.
Client connected.
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,155
Write to thread buffer: render
Write to thread buffer: blink 1,FFFFF,000000,100,3
Write to thread buffer: render
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,255
Write to thread buffer: render
Thread stop.
Running thread.
Waiting for client to connect.
Enter thread 1,0,110.
fill 0,0,0,13,0
Changing brightness 0, 155, 0, 13
Render (null)
blink 0, 15794175, 0, 100, 3, 0, 13
Render (null)
fill 0,0,0,13,0
Changing brightness 0, 255, 0, 13
Render (null)
Exit thread.
Client connected.
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,155
Write to thread buffer: render
Write to thread buffer: blink 1,FFFFF,000000,100,3
Write to thread buffer: render
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,255
Write to thread buffer: render
Thread stop.
Running thread.
Error creating new thread: 11Waiting for client to connect.

Bigdeath, 2020-07-02:
Today the same after Stream:

Waiting for client to connect.
Enter thread 1,0,110.
fill 0,0,0,13,0
Changing brightness 0, 155, 0, 13
Render (null)
blink 0, 15794175, 0, 100, 3, 0, 13
Render (null)
fill 0,0,0,13,0
Changing brightness 0, 255, 0, 13
Render (null)
Exit thread.
Client connected.
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,155
Write to thread buffer: render
Write to thread buffer: blink 1,FFFFF,000000,100,3
Write to thread buffer: render
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,255
Write to thread buffer: render
Thread stop.
Running thread.
Error creating new thread: 11Waiting for client to connect.
Client connected.
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,155
Write to thread buffer: render
Write to thread buffer: blink 1,FFFFF,000000,100,3
Write to thread buffer: render
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,255
Write to thread buffer: render
Thread stop.
Running thread.
Waiting for client to connect.
Enter thread 1,0,110.
fill 0,0,0,13,0
Changing brightness 0, 155, 0, 13
Render (null)
blink 0, 15794175, 0, 100, 3, 0, 13
Render (null)
fill 0,0,0,13,0
Changing brightness 0, 255, 0, 13
Render (null)
Exit thread.
Client connected.
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,155
Write to thread buffer: render
Write to thread buffer: blink 1,FFFFF,000000,100,3
Write to thread buffer: render
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,255
Write to thread buffer: render
Thread stop.
Running thread.
Error creating new thread: 11Waiting for client to connect.

tom-2015, 2020-07-03:
Can you check how many threads there are running in the ws2812-svr?
You can find this with the command:

`ps -o nlwp <PID>`

with <PID> the process ID of the ws2812-svr (you can find the process ID with the command TOP, number in the first column at the row of the ws2812-svr command).

Also check the total number of threads with the command:

`ps -eo nlwp | tail -n +2 | awk '{ num_threads += $1 } END { print num_threads }'`

tom-2015, 2020-07-03:
Also try the current version fixed something when too many sockets connected but this did not generate the same error on my Pi as you have.

Bigdeath, 2020-07-03:
Have installed it. After my stream i give you a feedback. I think it will be tomorrow afternoon or Sunday. thx thx thx!

Bigdeath, 2020-07-03:
Now im in my stream and this is my output:

ps -eo nlwp | tail -n +2 | awk '{ num_threads += $1 } END { print num_threads }':
137 Threads
---------
root@BigPi2:/home/pi# ps -o nlwp 19999
NLWP
   1
Memory: 75.7/929M
CPU: Load: 0.06 0.06 0.00
Tasks 46, 12thr; 1 running

Waiting for client to connect.
Enter thread 1,0,110.
fill 0,0,0,13,0
Changing brightness 0, 155, 0, 13
Render (null)
blink 0, 15794175, 0, 100, 3, 0, 13
Render (null)
fill 0,0,0,13,0
Changing brightness 0, 255, 0, 13
Render (null)
Exit thread.
Client connected.
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,155
Write to thread buffer: render
Write to thread buffer: blink 1,FFFFF,000000,100,3
Write to thread buffer: render
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,255
Write to thread buffer: render
Thread stop.
Running thread.
Error creating new thread: 11Cannot allocate memory
Waiting for client to connect.
Client connected.
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,155
Write to thread buffer: render
Write to thread buffer: blink 1,FFFFF,000000,100,3
Write to thread buffer: render
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,255
Write to thread buffer: render
Thread stop.
Running thread.
Waiting for client to connect.
Enter thread 1,0,110.
fill 0,0,0,13,0
Changing brightness 0, 155, 0, 13
Render (null)
blink 0, 15794175, 0, 100, 3, 0, 13
Render (null)
fill 0,0,0,13,0
Changing brightness 0, 255, 0, 13
Render (null)
Exit thread.
Client connected.
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,155
Write to thread buffer: render
Write to thread buffer: blink 1,FFFFF,000000,100,3
Write to thread buffer: render
Write to thread buffer: fill 1,000000
Write to thread buffer: brightness 1,255
Write to thread buffer: render
Thread stop.
Running thread.
Error creating new thread: 11Cannot allocate memory
Waiting for client to connect.

"Cannot allocate memory" Makes no sense.. Enough Memory left.

tom-2015, 2020-07-04:
I've fixed another memory leak today, can you check current version?
Also added command set_thread_exit_type which can be used to set if server should wait for execution of the current thread before starting a new script at next client connect / terminate current script and immediately launch next.

Bigdeath, 2020-07-04:
Oh Sounds Great! I was hoping for something like that.

Bigdeath, 2020-07-05:
Now after 2 Streams and put the "set_thread_exit_type" in every Script, i have no bug anymore.

But one time a older Bug happens. Have 2 Server on my Pi. One for Twitch, One for 4 LEDs inside the pi. 

The Twitch LEDs have a 14 LED Stripe. Inside pi only 4 LEDs. One time after 9h Stream the Effekt for the 4 LEDs inside the pi were suddenly on the 14 LED stripe. And the Effect for the 14 LED stripe suddenly on the 4 LED stripe. I hope you understand my bad english.

Is that a Bug from Pi itself? Have no Debug Info oder fails inside the servers.

Bigdeath, 2020-07-06:
This is the script for server 2 and the 4 LEDs:

<?php
send_to_leds("
thread_start
brightness 2,255
blink 2,FFFFFF,000000,100,10
render 2
rainbow 2
render 2
do
       rotate 2
       render 2
       delay 1000
loop
brightness 2,255
fill 2, 000000
render 2
thread_stop
");
function send_to_leds ($data){
   $sock = fsockopen("192.168.178.29", 9997);
   fwrite($sock, $data);
   fclose($sock);
}
$timestamp = time();
$datum = date("d.m.Y - H:i:s", $timestamp);
echo $datum;
$datei = fopen("./triggers.txt", "a+");
$date=date("d.m.Y - H:i:s");
fwrite($datei, $daten."LEDPION: $date\r\n");
fclose($datei);
?>

I will try to explain it again and in more detail.

Server 1 Twitch (14 LEDs)

Server 2 case Raspi (4 LEDs)

Suddenly the effect (script see above) of server 2 appears on the first 4 LEDs of server 1. Mostly this happens when the chat script for server 1 (see comment June 29th) is sent. Then i have on the 14 LED Stripe from LED 1-4 the Effect from Server 2 and from LED 5-14 all white. I have not had the problem often, but I have already had it 20 times in total. The servers suddenly change the GPIO. Scripts that are sent to server 1 are then displayed on server 2 or the LEDs on server2. I assume this is a RasPi bug?
When i only restart Server 1 then the effect jumps bag to Server 2 and perform by itself.

tom-2015, 2020-07-06:
It's probably something in the code that drives the DMA controller. But unfortunately it's not my code...
In future I may implement 2 threads in 1 server that should fix it.

Bigdeath, 2020-07-07:
Ok. Today i had the problem 2 times in 6h stream. But then i wait for a fix :-) thx!

tom-2015, 2020-07-10:
Something you can try while waiting is using a different DMA channel for both servers.
This can be done with the init command:
`init 800000,9`
the other server:
`init 800000,10`

Be careful with changing the DMA number (default 10) because some channels are used by the OS (like channel 5 for SD card). If you use wrong number it can corrupt SD card and crash OS (so backup the SD before you try). I don't know how to find out which DMA number is already in use...

Bigdeath, 2021-06-07:
1 year later, is there any progress?

tom-2015, 2021-06-08:
Did you try the latest version? There is no need to have 2 servers running anymore as you can start multiple threads from the main TCP connection.

```
thread_start 1
code for led strip 1
thread_stop
```

```
thread_start 2
code for the led strip 2
thread_stop
```

There are commands to stop executing threads / wait for thread to complete etc...

Bigdeath, 2021-06-13:
Jeah i installed now the new Version. Both working. :-) Now I'll wait and see if the same problems arise again.

Bigdeath, 2021-06-23:
Now it works over a Week without Problems. I think it solves. Thx!

closed 2021-06-23 by Bigdeath