tom-2015/rpi-ws2812-server #44: Cannot get a daemon to use port 9999

hans-martijn, 2020-05-12:
Hello,

my /etc/ws2812svr.conf is:

_#mode must be first setting in file!
mode=tcp
port=9999
file=/home/pi/test.txt
pipe=/dev/leds
init=setup 1,20,4,0,255,12;init_

if I start my server and I do a netstat -plnt I get any port above 35000, different every time. I probably have an error somewhere, but I cannot seem to find it. The server itself works, so that's not the problem.

Thanks in advance,
Hans

Bigdeath, 2020-05-12:
Start it with root privileges.

hans-martijn, 2020-05-12:
How else can I start a daemon?

Bigdeath, 2020-05-12:
sudo

hans-martijn, 2020-05-12:
I know, I wouldn't know how else to start a daemon...
[image]

jQubz, 2020-05-12:
I think I fixed this in https://github.com/tom-2015/rpi-ws2812-server/pull/45

The daemon was always trying to listen on port 0 :)

closed 2020-05-13 by tom-2015

hans-martijn, 2020-05-13:
That works perfectly, thanks!!!