tom-2015/rpi-ws2812-server #46: TCP command produces different outputs.

BloodSKreaper, 2020-05-26:
I am currently developing a smarthphone app for my LED-Strip and the rpi-ws2812-server is a good serverside basis to build on (only works in local network).

One of the things I want to implement is an animation that works as follows:

every third LED gets lid on an then the LEDS are being rotated one to the right, little pause, again rotation, pause, rotating, ....

I came up with the following code:

`thread_start
  do 
    fill 1, FF0000, {0}, 1
  loop 268, 3
  do
    rotate 1,1,2
    render
    delay 200
  loop
thread_stop`

However this only works as expected the first time the command is being sent to the server. After stopping the thread by connecting another time via tcp the effect is stopping as intended.

When sending the command again, the first loop is being ignored and thus the LED wont get filled (except the first one). The second loop is being executed as intented.

So for some reason the server is being ignoring to loop.

Hope you can help me fix this.

Kind Regards
BloodSKreaper

Bigdeath, 2020-06-10:
To prevend much problems i use at any TCP command at start:

fill 1,000000
brightness 1,255
render

and at end: 

fill 1,000000
brightness 1,255
render

Thats my Solution and works great.

tom-2015, 2020-06-14:
Can't reproduce the problem at this moment but did you add a newline (or a ;) after the "thread_stop" command?

Can you download & install the latest version please?
If it doesn't fix the problem try running the program from a command window with the -d (debug option) and post output here.

BloodSKreaper, 2020-06-14:
Updating solved the problem.

Thank you for your efforts!

closed 2020-06-14 by BloodSKreaper