tom-2015/rpi-ws2812-server #48: chaser crashes the daemon

hans-martijn, 2020-07-05:
If I try to send a chaser command to the daemon, it crashes and only works again after restarting the daemon. It's not really a problem (I don't really need it), but I might do something wrong? This is de line I send to the daemon:

$data .= "chaser 1,0,$color,0,20,0,2,$brightness,0;";

Or actually, it's the code in my php page.

Thanks in advance and kind regards,
Hans

Bigdeath, 2020-07-06:
Yes chaser, crashes my servers too.

Debug: 
Enter thread 1,0,33.
chaser 0 0 48 0 20 0 2 11 0 0

My test:

`
<?php
send_to_leds("
thread_start
chaser 1,0,FFFFFF,0,20,0,2,255,0
thread_stop
");
function send_to_leds ($data){
   $sock = fsockopen("192.168.178.29", 9999);
   fwrite($sock, $data);
   fclose($sock);
}
`

Nothing perform.

tom-2015, 2020-07-10:
There was an argument missing in the documentation, I've updated it today.
Try:
`$data .= "chaser 1,0,$color,0,**1**,20,0,2,$brightness,0;";`

closed 2021-02-28 by hans-martijn