tom-2015/rpi-ws2812-server #49: readpng with start position >= 963 not work

CBMOD, 2020-07-16:
To localize the error I tested with a 1 pixel image

readpng 1,/home/pi/ledwall/1pixel.png,000000,962,2000,0,=,0; >> works
readpng 1,/home/pi/ledwall/1pixel.png,000000,963,2000,0,=,0; >> not works

tom-2015, 2020-07-17:
Can you change led 963 (with fill command or something else)?

CBMOD, 2020-07-19:
i testet:

fill 1,ff0000,963,1
fill 1,ff0000,1925,1

rotate

the commands fill and rotate also work perfectly for addresses larger than 962

tom-2015, 2020-07-19:
can you try running with debug parameter (-d) and post output?
(stop deamon if you have it running: sudo service ws2812svr stop)
sudo ./ws2812svr -d

also try this command: readpng 1,/home/pi/ledwall/1pixel.png,000000,963,1,0,=,0;

CBMOD, 2020-07-22:
Debug:

root@LEDPANEL1:/home/pi/rpi-ws2812-server# sudo ./ws2812svr -d
setup 1,1925,9
Initialize channel 0,1925,9,0,255,18

init
Init ws2811 800000,10

readpng 1,/home/pi/ledwall/1pixel.png,000000,962,2000,0,=,0;          >>work
readpng 0,/home/pi/ledwall/1pixel.png,0,962,963,0,0,0
led 962= r 249,g 178,b 180,a 0, PNG channels=3, PNG idx=0

render
Render (null)

readpng 1,/home/pi/ledwall/1pixel.png,000000,963,2000,0,=,0;          >> not work
readpng 0,/home/pi/ledwall/1pixel.png,0,963,962,0,0,0
led 963= r 249,g 178,b 180,a 0, PNG channels=3, PNG idx=0

render
Render (null)

fill 1,F9B2B4,963,1          >> work
fill 0,11842297,963,1,0

render
Render (null)

tom-2015, 2020-07-22:
I found something, can you try current version and see if it's fixed?

CBMOD, 2020-07-26:
it's fixed - good work!

closed 2020-07-26 by CBMOD