tom-2015/rpi-ws2812-server #51: How do you invert the logic?

leoheck, 2020-09-01:
Does invert the logic work? I could not see any example here.

leoheck, 2020-09-01:
I am driving my LEDs with a PMOS transistor, so they now they are active low. Does this tool support this inversion? What I can do to invert the whole output logic?

tom-2015, 2020-09-02:
Yes you can set it in the setup command see the invert parameter after led type.
```
    setup 
		<channel>, 						#channel number
		<led_count>, 					#number of leds in channel
		<led_type>, 					#type of led (3 color or 4 color) default 0
		<invert>, 						#invert output, default 0
		<global_brightness>, 			#global brightness level for channel (0-255), default 255
		<gpionum>						#GPIO output number, default 18 for more see 'GPIO usage' at this page: 
```
example:

setup 1,10,0,1

If it doesn't work let me know then there is a bug...

leoheck, 2020-09-04:
Yeah, I made new hardware (active low) that was not with me at that time. It is different than my original setup which was active high. I was trying to use it for the first time remotely through a webcam. Which is far from the ideal. I was using the code exactly as you said before. And nothing was working.

My setup is this
```
setup 1,8,3,1,255,18
```

Now I have the board with me. Then I will test and let you know if I found any issue with the software.

leoheck, 2020-09-04:
So, this is my first test.

**(1) My routine to turn all of them on white. Nothing happens.**
It does nothing including after running (2).

```
setup 1,8,3
init

fill 1,ffffff
brightness 1,255,0,8

render
```

_Executing it with debug flag enabled_
```bash
sudo ./ws2812svr -d -f patterns/hb_white.txt 
Opening patterns/hb_white.txt.Initialize channel 0,8,3,0,255,18
Init ws2811 800000,5
fill 0,16777215,0,8,0
Changing brightness 0, 255, 0, 8
Render (null)
```

**(2) My routine to turn them off. Is turning the 1st one green.**
```
setup 1,8,3,1
init
fill 1,000000,0,8 # off
render
```

_Executing it with debug flag enabled_
```bash
sudo ./ws2812svr -d -f patterns/hb_off.txt
Opening patterns/hb_off.txt.Initialize channel 0,8,3,1,255,18
Init ws2811 800000,5
fill 0,0,0,8,0
Render (null)
```

leoheck, 2020-09-05:
Ok, so I was testing jgarff/rpi_ws281x repo, and it didn't work too.

After I bypassed my level-shifter/inverter using the active high code it worked. So, my hardware is good and working.
The issue is the inverting code now.

I found this issue, that is maybe related, but it is kind of hard to understand what he is doing. No one has answered him anyway https://github.com/jgarff/rpi_ws281x/issues/337

leoheck, 2020-09-05:
I started to see light and colors when I changed this PWM initialization. They are not reproducing the pattern yet, but now I can see something working which is pretty nice.

[image]

My inverting level-shifter from the Raspberry 3.3v to the 5v leds inputs is this one.
[image]

leoheck, 2020-09-05:
Yeah, that pull-down in front of the Q1 is not good.
The code is working and inverting just fine. Thanks for your help @tom-2015