tom-2015/rpi-ws2812-server #52: Error set SO_KEEPALIVE

leoheck, 2020-10-14:
Hi, I am seeing many `Error set SO_KEEPALIVE` messages using as a server as in `ws2812svr -tcp`

tom-2015, 2020-10-14:
I've changed the function to perror which should print some more information why it is failing. Can you check the current version and if you get information why it is failing?

leoheck, 2020-10-14:
Sure. I am working on a fork but it was not changed I will check if it is updated with the last version from here.
I also saw another issue here https://github.com/tom-2015/rpi-ws2812-server/issues/7 with the same subject.
There you said to add a small delay between writes. I am trying this but I am not sure If it will work since I don't think I am wrinting on this device so much.