tom-2015/rpi-ws2812-server #57: Is there a way to make the LED's react to music input?

Loxitation, 2021-01-30:
Hey, as the title says, is there, or will there be a way? 👯 :)

tom-2015, 2021-02-01:
At this moment it's not implemented in the program, main reason is I don't know how to convert audio into commands to change the led color/brightness :(.

Bigdeath, 2021-02-01:
Is there a way to convert the Mic Signal into a wave Form? Then for high, mid and Bass Set colors and fade them random. I hope you understand what I mean. It where good to know how other products work with Sound 2 Light 🤔

Loxitation, 2021-02-01:
@tom-2015 What if you don't have to convert the audio into commands? I mean the audio signal needs to be processed, thats right. But depending on the amplitude of the track, it can send or it will send "values" or a graph. Your part would be to grab parts of the path every x ms and read a value, lets say 1-10. 1 = low, nothing happens, maybe little voice part, where 10 would be the base kick while the song drops. Now to transfer that into a command you could use if(value=10){DoStuff}. Audio Input for the pi could come trough cheap usb soundcards, I'll google a bit in my spare time, maybe I can come up with some more help :/ Thats the only feature I'm missing, being able to run my "shows" sound activated 👯

tom-2015, 2021-02-02:
🤔you mean make the brightness change with amplitude of audio signal?

Loxitation, 2021-02-02:
$499.99, definetly not :D

Loxitation, 2021-02-02:
Is this helpfull? https://stackoverflow.com/questions/36250228/read-frequency-from-mic-on-raspberry-pi

Loxitation, 2021-02-04:
https://github.com/scottlawsonbc/audio-reactive-led-strip the dsp.py should be interesting. Its the digital signal processing.

tom-2015, 2021-02-04:
Yes that can be interesting but will take some time to implement.

Bigdeath, 2021-06-06:
Any Updates? Possible?

tom-2015, 2021-06-08:
Didn't have time to start with audio support.
Now finishing 2D matrix support first, after this I'll take a look at audio input.

tom-2015, 2021-07-19:
Release 6.1 has 3 new functions for audio effects.
It's still a beginning and the DSP is probably far from perfect.