tom-2015/rpi-ws2812-server #58: Is there a way to read a .txt file when the program is already launched?

sydneybeaulieu, 2021-02-01:
I know that to read a file I have to do ./ws2812svr -f text_file.txt , but can I read a file after I have run the init command

Loxitation, 2021-02-01:
sure you can, thats what I do in my shows. Doing it with php files.

After reboot my Pi inits the led's, then waits. 

My textfile is idle.txt:
`thread_start
set_thread_exit_type 0,0
fade 1,255,0,50,10,0
render
rainbow 1,1,0,40,0,223
render
fade 1,0,100,50,10,0
brightness 1,100
render
do
    rotate 1,1
    render
    delay 20
loop
render
thread_stop`

and my php looks like this:

`<?php

send_to_leds("
thread_start
set_thread_exit_type 0,1
fill 1,000000
render
brightness 1,100
rainbow 1,1,0,255,0,223
render
do
       rotate
       render
       delay 20
loop 400

fill 1,000000
brightness 1,000
render
thread_stop
");
function send_to_leds ($data){
   $sock = fsockopen("192.168.178.53", 9999);
   fwrite($sock, $data);
   sleep(9);
   fclose($sock);
   
   $filename = "idle.txt";
   $fp = fopen($filename, "r");
   $content = fread($fp, filesize($filename));

   start_idle("$content
   ");
}

function start_idle($data){
   $sock = fsockopen("192.168.178.53", 9999);
   fwrite($sock, $data);
   fclose($sock);
}
?>`

the 192.168.178.53 is my pi. This code starts a rainbow effect, rotates it for 9 secs, then loads in the idle.txt and proceeds with it.

tom-2015, 2021-02-02:
If you don't want to use PHP start the program with -p flag to create a named pipe:

sudo ./ws2812svr -p /dev/ws281x

This will create a file /dev/ws281x  reading a file and sending the contents to /dev/ws281x:

cat /home/pi/my_commands.txt > /dev/ws281x

replace /home/pi/my_commands.txt with the location of your file containing commands

sydneybeaulieu, 2021-02-02:
okok thank you a lot

closed 2021-02-02 by sydneybeaulieu