tom-2015/rpi-ws2812-server #59: make install fails to disable missing service first time

zacaj, 2021-02-04:
first time install on a fresh rpi, ran all the commands in the readme in order.  when I got to `sudo make install`, it failed with

    systemctl stop ws2812svr.service
    Failed to stop ws2812svr.service: Unit ws2812svr.service not loaded.
    make: *** [makefile:58: install] Error 5

commended out that line in the makefile, everything worked fine, so I assume it's just erroring out when it should be ignoring the error, or something similar

tom-2015, 2021-02-04:
yes it should ignore, I'll try to fix that

KHome, 2021-02-19:
Hi, had the same issue, too.
Solved by following steps:
- copy` ws2812svr.service` from git repository to `etc/systemd/system/`
- execute `sudo make install` then works fine

Result:
```
systemctl stop ws2812svr.service
cp ws2812svr.service  /etc/systemd/system/ws2812svr.service
cp -n ws2812svr.conf /etc/ws2812svr.conf
cp ws2812svr /usr/local/bin
systemctl daemon-reload
systemctl stop ws2812svr.service
systemctl enable ws2812svr.service
Created symlink /etc/systemd/system/multi-user.target.wants/ws2812svr.service → /etc/systemd/system/ws2812svr.service.
systemctl start ws2812svr.service
```

@tom-2015 : To reproduce this issue following file has to be removed before: `/etc/systemd/system/ws2812svr.service`