tom-2015/rpi-ws2812-server #60: when connecting 3 SK6812, deamon gives error

hans-martijn, 2021-02-28:
Hi,

I was trying to connect 3 SK6812-rings to one pi, as in the text example in the code. I got the following error from the deamon:

ws2812svr[883]: Error no free channels anymore for led type WS2811/SK6812, max number of PWM led strings is 2

Did I do something wrong?

Kind regards,
Hans

tom-2015, 2021-02-28:
You can only connect 2 SK6812-rings to the same Raspberry. If you want to connect more you can try to connect the output of ring 2 to input of ring 3 of course you must adjust the commands for ring 3 to start at the right offset in led string 2.

hans-martijn, 2021-02-28:
Thanks Tom,

Since I already built the lights in my boat, I'll simply buy a new raspberry I guess. Thanks for your quick answer!!!

closed 2021-02-28 by hans-martijn