tom-2015/rpi-ws2812-server #61: Weird BRG instead of RGB

Timborin0, 2021-03-11:
Hi,
if I try the `fill` function like `fill 1,FF0000` I'm getting blue instead of red.
Also if I'm trying to fill with this BRG hex color `#FF6400`, I'm getting a pink color instead of cyan, also seems like that this BRG is flipped/reversed? <- If I try with the BRG hex `#FF34FF` I'm getting the cyan.

I did everything the author had in the `README.md` for the basic test setup.
I removed the folder and ran `make` again but still didnt got me where I wanna be.
Im using a RaspberryPi ZeroW.

hans-martijn, 2021-03-12:
Hi Tim,

If you check under the setup section of the readme, you can several initializations for different LED types. If you are unsure which one you have (RGB, GRB, GBR, etc.) it's a trial and error test to find the correct one. Try a different initialization and then try to fill the different colors.

Even when it sells as RGB, that doesn't mean it is RGB, especially on a big Chinese website.

-Hans

Timborin0, 2021-03-13:
Hi,
thanks for your reply!
You were right.
Shouldve thought of this myself... and I thought my strip is RGB but apperantly it wasnt.

closed 2021-03-13 by Timborin0