tom-2015/rpi-ws2812-server #62: Multiple PHP

Chlorex, 2021-05-19:
Hello everybody.

So far I have understood the execution with "./ws2812svr -f file.txt".
I wrote a web server including PHP script and buttons so that I can start various programs via the website.

Now to my problem:
Server starts with
"sudo ./ws2812svr -tcp 9999"
The first (no matter which) PHP file is executed, but the second PHP file is not.

Didn't find a stop or reset function.

Is there a "Next PHP File" function included?

Greetings from Germany, great job.

My example:
`<?php 
send_to_leds("
	reset;
	setup 1,299,3;
	init;
	chaser 1,0,00ff00,1,1,10,220,50;
	");

function send_to_leds ($data) {
	$sock = fsockopen("127.0.0.1", 9999);
	fwrite($sock, $data);
	fclose($sock);
}

?>`

tom-2015, 2021-05-20:
I'm not sure if I understand your question but I guess you want to abort/stop the current command next time you run the PHP script? This is the solution:

Start the server with the setup/init command:

`sudo ./ws2812svr -tcp 9999 -i "setup 1,299,3; init;"`

Now with PHP you send commands to create a background thread and start the chaser in this thread:

```
send_to_leds("thread_start 1,0;
                              chaser 1,0,00ff00,1,1,10,220,50;
                     thread_stop;
                    ");
```

Next time you run the PHP code the chaser command will abort and your next command will start to run:

```
send_to_leds("thread_start 1,0;
                              <ENTER NEXT COMMANDS HERE>
                     thread_stop;
                    ");
```

Chlorex, 2021-05-20:
yes, this is the solution... 

I missed it : 
`thread_start with the value of a thread that is still running if join_type is 0 the thread will be aborted immediately`

thank you... 
Great job.

closed 2021-05-20 by Chlorex