tom-2015/rpi-ws2812-server #64: white leds on by draw_image with LED Typ 9 SK6812_STRIP_GBRW

CBMOD, 2021-08-08:
i have a file with 34x1 pixel, with used only colour red colour  and black
i used readpng 1,<file>;render;  >>> only red Led's on >>> OK
i used:

config_2D 1,34,1;
draw_image 1,<file>;render; (i used the same file) >>> by the red pixels are red and white Led's, by black Pixels are white Led's on

tom-2015, 2021-08-12:
The 2D functions are not optimized for use with the SK6812 RBGW chips. The white LEDs are on because it uses the Alpha (transparancy) value that is fully on (255) for the white LEDs.

If you only have a 32x1 pixel strip better use the 1D readpng function.

CBMOD, 2021-08-19:
it's only a test with the 1x34 stripe... ;-)
i remove the alpha channel - same problem

and the repeatet gif animation i can never stop it 
a new draw command dont overwrite the first draw command

tom-2015, 2021-08-19:
When drawing animated GIF use the max_loops parameter to specify how many times to repeat. Another option is to start the GIF in a different thread, you can then stop it by killing the thread.

Removing alpha from the image will not solve the problem. I've changed something in the main.c file that should prevent this problem. Can you try download and recompile see if it fixes this?

CBMOD, 2021-08-20:
it work's !! great job 🥇

closed 2021-08-20 by CBMOD