tom-2015/rpi-ws2812-server #65: First LED is constantly white

MarkusOsterBEC, 2021-09-30:
Hi, if I just use a fill with a constant colour the first led is a constant white. Am I doing something wrong with the configuration? How can I suppress this behaviour?

tom-2015, 2021-09-30:
Probably has something to do with initialization command. What if you tell the setup command there is 1 more LED, does this solve the problem? So if you have 100 leds, call setup 1,101,...

MarkusOsterBEC, 2021-10-01:
I just did a test with a smaller setup (12 LEDs) and the first two were behaving strangely. I used 

`setup 1,11,3,0,100,18;
init;
fill 1,FF0000;
render;`

And then I increased the count of the LEDs up to 12,13 and 14. In some cases the first and second LED where white (at a higher brightness than the rest) and after using a second render command the second LED blinkt between the right colour and white. 

Could it be a hardware problem with bad signal quality, even though the wire between the Raspberry PI is only around 10cm long? 

PS: I am using the ws2812 server as a service and then telnet to send the commands to the PI.

Bigdeath, 2021-10-01:
Try to use the same ground for pi and led. Otherwise put a 100Ohm resistor to the Signal cable.

tom-2015, 2021-10-01:
What voltage are using for the LEDs? The signal coming from the Pi is 3.3V, if you power LEDs with 5V you may need a level converter to bring the 3.3V output signal from the Pi to 5V [https://www.adafruit.com/product/1787](https://www.adafruit.com/product/1787).

MarkusOsterBEC, 2021-11-02:
@Bigdeath They already had the same ground. 

@tom-2015 Yes it was the wrong signal voltage, I now have a logic level shifter between the PI and the first LED and it works perfectly. Thank you for your help. 

Sorry it took so long, I had to wait for the parts to test this. 
I think this thread can now be closed.

closed 2021-11-08 by MarkusOsterBEC