tom-2015/rpi-ws2812-server #66: Problem with instalation

osnapus, 2021-11-06:
Hello, I have problem today (few days ago everything was without problems) with instalation few packages

Ex.
Błąd:1 http://raspbian.raspberrypi.org/raspbian buster/main armhf libjpeg62-turbo-dev armhf 1:1.5.2-2+b1
  404  Not Found [IP: 93.93.128.193 80]
Błąd:2 http://raspbian.raspberrypi.org/raspbian buster/main armhf libjpeg-dev all 1:1.5.2-2
  404  Not Found [IP: 93.93.128.193 80]
E: Nie udało się pobrać http://raspbian.raspberrypi.org/raspbian/pool/main/libj/libjpeg-turbo/libjpeg62-turbo-dev_1.5.2-2+b1_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Nie udało się pobrać http://raspbian.raspberrypi.org/raspbian/pool/main/libj/libjpeg-turbo/libjpeg-dev_1.5.2-2_all.deb  404  Not Found [IP: 93.93.128.193 80]
E: Nie udało się pobrać niektórych archiwów, proszę spróbować uruchomić apt-get update lub użyć opcji --fix-missing.

Someone know whats the problem?

tom-2015, 2021-11-07:
Did you try?
sudo apt-get update

Then install de dependencies.

osnapus, 2021-11-07:
Yes, I try, not work.

I can't find this files on server when I open address from links in error 
info...

tom-2015, 2021-11-07:
I've done a test with a new SD card and fresh install of the Raspberry Pi OS, no problems all dependencies installed.
It used this:
Get:24 http://mirror.as43289.net/raspbian/raspbian buster/main armhf libjpeg62-turbo-dev armhf 1:1.5.2-2+deb10u1 [181 kB]

Did you use the official Raspberry Pi OS software?

https://www.raspberrypi.com/software/operating-systems/

Can you try with a new SD card and fresh install?

osnapus, 2021-11-07:
Of course... I was tryi g install it on Raspbian from 2019 december, when i 
make MagicMirror. Maybe this is a problem, old OS without updates.

I will try, thank You so much for help.

Peter