tom-2015/rpi-ws2812-server #67: SK9822 > 127 LEDs throws error

timvis, 2021-12-04:
Hey Tom, I'm testing SK9822 Stripes with 144 LEDs on an RPI 1, but I'm getting unexpected results.

For a setup of 1,144,12 I get the Error "Can't send spi message /dev/spidev0.0sk9822 render failed on channel 0"

This gives me 127 red LEDs but also two white LEDs in the end:

```
setup 1,126,12
init
fill 1,ff
render 1
```

but this 63 red LEDs + one white LED:

```
setup 1,63,12
init
fill 1,ff
render 1
```

I suppose it's a bug, but maybe I also do something wrong. Thanks for looking.

tom-2015, 2021-12-05:
Can you try current version, I've change something but I'm not sure if it will fix the problem.
I don't have SK9822 around here a.t.m. to test.

If you still have the 'can't send spi message' try increasing the SPI buffer, see:
https://raspberrypi.stackexchange.com/questions/65595/spi-transfer-fails-with-buffer-size-greater-than-4096

timvis, 2021-12-05:
that fixed the problem. now it's working with 288 LEDs as expected at a first test. And thanks for the link to change the buffer.

tom-2015, 2021-12-05:
ok good to hear

closed 2021-12-05 by tom-2015