tom-2015/rpi-ws2812-server #68: Programm with threads is not executed

timvis, 2021-12-07:
Hey Tom, I run into a problem when I want to start a program where threads are defined.

` ./ws2812svr -f LED_program.txt
`

Nothing happens, **console output**:

```
Opening LED_program.txt.
READY
READY
```

**LED_program.txt**

```
setup 1,288,12;init;
brightness 1,0; fill 1,FFFFFF

thread_start 1
        #random_fade_in_out <channel>,<duration Sec>,<count>,<delay>,<step>,<sync_delay>,<inc_dec>,<brightness>,<start>,<len>,<color>
        random_fade_in_out 1,0,50,10,15,800,1,255,1,288,FF;
thread_stop 1

thread_start 2
        record_audio "plughw:CARD=Device,DEV=0"
        #light_organ <channel>,<color_mode>,<color(s)>,<color_change_delay>,<duration>,<delay>,<start>,<len>
        light_organ 1,3
thread_stop 2
```

Bigdeath, 2021-12-07:
Try it with "render" 😂

timvis, 2021-12-07:
Where? I dont get it... I want to overlay the light organ with the effect. This works without errors, has only one thread defined:

```
setup 1,288,12; init
brightness 1,0; fill 1,FFFFFF

thread_start 1
        random_fade_in_out 1,0,50,10,15,800,1,255,1,288,FF;
thread_stop 1

#thread_start 2
        record_audio "plughw:CARD=Device,DEV=0"
        light_organ 1,3
#thread_stop 2

render 1
```

Btw: Is there a clean exit option, when you CTRL-C ? Eg. reset all and send 000000 to the LEDs.

Bigdeath, 2021-12-07:
setup 1,288,12; init
brightness 1,0; fill 1,FFFFFF
RENDER 

thread_start 1
        random_fade_in_out 1,0,50,10,15,800,1,255,1,288,FF;
        RENDER
thread_stop 1

#thread_start 2 //why a # at beginning? 
        record_audio "plughw:CARD=Device,DEV=0"
        light_organ 1,3
        RENDER 
#thread_stop 2

it's possible to write render1 or render2
Try it. But for ever Operation you need a render command. Otherwise nothing happens.

I use this Server in screen. For stop and start I use screen commands. You can build a Script that sends all Black to the Server and kill the Server process.

tom-2015, 2021-12-07:
Can you check the commands work separately (without the threads)?

Also I'm afraid the combination will not work because the light_organ will always change the brightness of ALL LEDs no matter what the random_fade_in_out changed it to. For this to work you would need to multiply the brightness output of the light_organ command with the brightness of the random_fade_in_out which is not supported at this moment.