tom-2015/rpi-ws2812-server #70: Cairo crash ?

oscarmh, 2022-04-25:
Hi there I´m launching ws2812svr with this command (not running uin background)

./ws2812svr -i "setup 1,768,3;init;debug 2;brightness 1,32;config_2D 1,48,16,3,48,16;init_layer 1,0,1,0,1,0;text_input 1,0,8,48,8,6666,0,FF0000,000000,50,8,,,"enhanced_led_board-7.ttf""

very standard one and then I run a bash script like this 

#!/bin/bash
while :
do
        timedatectl set-timezone Europe/Berlin
        echo "   DUBLIN " | nc -w 0 127.0.0.1 6666
        date "+%H:%M:%S (%a %d/%m)"  | nc -w 0 127.0.0.1 6666
        #
        timedatectl set-timezone Europe/Paris
        echo "   MADRID " | nc -w 0 127.0.0.1 6666
        date "+%H:%M:%S (%a %d/%m)"  | nc -w 0 127.0.0.1 6666
        #
        timedatectl set-timezone America/New_York
        echo "  NEW YORK "  | nc -w 0 127.0.0.1 6666
        date "+%H:%M:%S (%a %d/%m)"  | nc -w 0 127.0.0.1 6666
        #
done

it is running in a RPI Zero but after 5 minutes or so, I got an error and crash. Any suggestions ?

oscarmh, 2022-04-25:
random... between 2 or 3 minutes

tom-2015, 2022-04-26:
Maybe as work around you can try to run the ./ws2812svr command in a separate bash file with a loop so it will auto restart after a crash?

oscarmh, 2022-04-29:
that is curious when I launch it I need to press intro so it activates the port 6666... have your tested it ?

commit b2d180d2 (2022-06-06): Fix bug in text_input

Fix bug in text_input
closes #70
closed 2022-06-06 by tom-2015