tom-2015/rpi-ws2812-server #71: Layer not working ?

oscarmh, 2022-05-22:
Hi team... what is wrong with this ? I cannot see the color... any help is appreciated 

setup 1,1536,3;
init;
brightness 1,20;
config_2D 1,96,16,3,16,16;
init_layer 1,1,1,0,1,0;
change_layer 1,1;
draw_image 1,red-green.png;
init_layer 1,2,1,0,1,0;
change_layer 1,2;
text_input 1,0,12,96,12,6666,0,0048BA,,10,12,,,enhanced_led_board-7.ttf

tom-2015, 2022-05-22:
did you do a "render;" command at the end?

oscarmh, 2022-05-22:
no success, no matter if I do render at the end or in other positions even with chanel signaling.

tom-2015, 2022-06-05:
I had some time today to take a closer look and I found 1 problem: the render operation of layer 2 is probably not correct:
- use operator 8 (CAIRO_OPERATOR_DEST_IN) if you want the background image to appear where text is
- use operator 2 (CAIRO_OPERATOR_OVER) if you want the text to be rendered on top of the image

Also make sure the red-green.png is available, normally it is in the examples folder

So this will give:

setup 1,1536,3;
init;
brightness 1,20;
config_2D 1,96,16,3,16,16;
init_layer 1,1,1,0,1,0;
change_layer 1,1;
draw_image 1,"examples/red-green.png";
init_layer 1,2,**8**,0,1,0;
change_layer 1,2;
text_input 1,0,12,96,12,6666,0,0048BA,,10,12,,,enhanced_led_board-7.ttf

OR

setup 1,1536,3;
init;
brightness 1,20;
config_2D 1,96,16,3,16,16;
init_layer 1,1,1,0,1,0;
change_layer 1,1;
draw_image 1,"examples/red-green.png";
init_layer 1,2,**2**,0,1,0;
change_layer 1,2;
text_input 1,0,12,96,12,6666,0,0048BA,,10,12,,,enhanced_led_board-7.ttf