tom-2015/rpi-ws2812-server #75: Raspberry Pi Zero 2 W Freezes at `init`

iv-mexx, 2023-05-08:
Hi there, right away I'm sorry for probably being too vague but I dont know how to get more useful information...

Everything worked fine until yesterday when I ran `apt upgrade` on my Raspberry Pi Zero 2 W. 
After that, as soon as `init` is called in `ws2812svr`, the Pi freezes and gets really hot.
I'm only connected via SSH and at that moment the SSH connection breaks, so I dont know how to get any more information...

When I put the same SD card into my older Raspberry Pi Zero 1, everything still works as expected....

tom-2015, 2023-05-09:
I once had that problem too and found it was a short circuit between 5V and 3.3V (soldering mistake). The Pi seemed to work perfect in the first place but always crashed when opening the ws2812svr. Later I found out some other stuff wasn't working anymore, SD card was very hot and then noticed the solder problem.

This might be a total different problem... What you can do is try using a different DMA channel but be careful as this may corrupt the SD card or do unexpected things.

Try if this library works: https://github.com/jgarff/rpi_ws281x which the ws2812svr is based on.

tstalzer, 2024-03-10:
Hello
We have the same issue with a Pi Zero 2W with the latest Raspbian image from the raspberry pi website.
As soon as we do an "init" after "setup 1,20", the system hangs
We triplechecked the connections - no short circuit on the board
Any ideas?
Kind regards
--Thomas

iv-mexx, 2024-03-10:
@tstalzer I was never able to resolve this, but I have since tried with multiple other Raspberry Pi Zero 2W and all of them had this issue. 
Paradoxically, I was also not able to find any older Raspbian image where it's working even though I know it was working before upgrading a day before opening this issue...